glenn20/micropython-espnow-images #1: Ready steady ......?

rubydesign, 2021-04-17:
Hi Glen,
so i got my (first) esp, installed python, fiddled 2 minutes and installed your esp-now image.
Great. And really, thanks for making those available, i would have no clue how to do that compilation. And doing all the work.

So now i am ready to try it out. But either there was no link to docs, or i didn't see, in which case i am sorry.

glenn20, 2021-04-18:
I have provided docs for the espnow in my PR at https://github.com/micropython/micropython/pull/6515 (see the link to readthedocs,io in the first post.) General micropython docs are at https://docs.micropython.org/en/latest/reference/index.html.

Good luck ;-).

closed 2021-04-18 by glenn20

rubydesign, 2021-04-18:
thanks, sorry i missed it in the readme

glenn20, 2021-04-18:
I just added it to the readme - in response to this issue. Thanks for raising it ;-).

commit 470cca2b (2021-04-18): Update README.md

Add link to espnow and micropython docs. (Closes #1).