glenn20/micropython-espnow-images #3: ESPNow.irecv() returns incorrect MAC address

philwilkinson40, 2021-05-25:
G'day Glenn, I have just started messing around with your espnow binaries and hit a problem.

software 20210511_espnow-g20_v115 firmware-esp8266-GENERIC.bin 
hardware is a bunch of Lolin Wemos D1 Minis 

I am sending messages and receiving messages using your examples. The messages are sent and received fine.
 
However the ESPNow.irecv() returns a MAC bytearray of the sender as
b'\xdeO"\x1c\x88l'
however the MAC address of the sender is actually
b'\xdcO"\x1c\x88l'

another example was
b'\xdeO"\x1c\x8a\xc8'
when the MAC address of the sender was actually
b'\xdcO"\x1c\x8a\xc8'

the above indicates a single bit is in error.

glenn20, 2021-05-27:
Hi Phil,

I think you are seeing the sometimes confusing aspect of wifi on the ESP platform. The esp8266 (and esp32) have two wifi interfaces. In your case the `network.WLAN(0).config('mac')` with show the mac address of the STA_IF interface as `b'\xdcO"\x1c\x88l'` and `network.WLAN(1).config('mac')` with show the mac address of the AP_IF interface as `b'\xdeO"\x1c\x88l'`. On the ESP8266 the two interfaces always differ by 2 in the first octet - on the esp32 it is in the last octet.

I am guessing that your esp8266 has been configured so that the AP_IF interface is the active interface for sending espnow packets. If you want to change that, you can `network.WLAN(1).active(False) ; network.WLAN(0).active(True)`. (Also - remember that the esp8266 remembers it's last wifi config when it reboots.)

If that is not the case, let me know and I will follow up the report in more detail.

philwilkinson40, 2021-05-27:
Beautiful! that does the trick.
I can assure you there is no way I would have found that in the ESP docs, thanks @glenn20 !
Given I am possibly not the first or the last person to overlook that detail, is it worth putting a note in the docs?

closed 2021-05-27 by philwilkinson40

glenn20, 2021-05-27:
Yes, it took me some time of reading and trial and error to work out the
subtleties of networking on the esp platforms. I didn't want to have the
micropython espnow docs be full of clarifications and explanations of
general esp network management, but I find myself adding more and more such
clarifications ;-). I will put something in the next version.