glenn20/micropython-espnow-images #4: ESPNow and WiFi - implementing ESPNow patches

philwilkinson40, 2021-07-03:
Glenn,
I am just trying to interpret your [explanatory notes on ESPNow and WiFi](https://micropython-glenn20.readthedocs.io/en/latest/library/espnow.html#espnow-and-wifi).
I am using two ESP32s to send and receive both using your [latest pre-compiled binaries for the esp32](https://github.com/glenn20/micropython-espnow-images/tree/main/20210511_espnow-g20_v115). Everything works fine sending and receiving ESPNow data on AP_IF and STA_IF. Problems arise when the recieving ESP32 is also connected to WIFI on STA_IF.
I am not able to either disable the power-saving mode on the STA_IF interface or configure the AP_IF interface using a channel. Both throw error exceptions or brownouts.
I assume this is due to the points you made in the text:

- (related to disabling the power-saving mode on the STA_IF interface) **This requires the ESPNow patches on ESP32 (not supported in micropython v1.15).** 
- (related to using the AP_IF interface for ESPNow messages)  **Some versions of the ESP IDF don’t permit setting the channel of the STA_IF at all, other than by connecting to an Access Point (This seems to be fixed in IDF 4+). Micropython versions without the ESPNow patches also disallow setting the channel of the STA_IF.**

I just wanted to understand whether your pre-compiled binaries implement the  patches or whether we need to compile our own binaries from source?
thanks for any clarification,
Phil

glenn20, 2021-07-12:
Hi Phil,

Sorry for the delay - been away from home till yesterday. The 20210511 pre-compiled binaries do include the required patches to the ESP32 wifi code to support the 'ps_mode' option and they have been built against ESP-IDF v4.2.1 (which should work fine). So they should include everything you need (including ability to set the channel on STA_IF).

It seems you are trying to use the espnow module on STA_IF while connected to a wifi AP on STA_IF. In this case, you have no choice of channel, you must use the channel the AP uses - so you shouldn't set the channel at all. Also remember to call `config(ps_mode=network.WIFI_PS_NONE)` **after** you have connected to the AP.

This is working for me and the small range of devices I have at my disposal. If it is not working for you - let me know.

Also - if you are seeing brownouts, you might want to check your ESP32 is getting enough power - the wifi consumes quite a bit of extra power - and gets quite warm with WIFI_PS_NONE.

glenn20, 2021-07-12:
Also, if you are getting exceptions, send the details and I'll look into it.

philwilkinson40, 2021-07-13:
Thanks again for the follow up Glenn.
Still no luck on your Option 1; disabling power saving.  No exceptions or brownouts, just nothing received. See truncated scripts below. I have double/triple checked mac addresses. 
Option 2, using AP_IF. throws a guru meditation error when I try to select channel selection.
By the way, your Option 3 (Configure ESPNow clients to retry sending messages) is running a treat for a dozen or so sensor nodes around my house.

```python
#sender node
import network
import config
from esp import espnow

w0 = network.WLAN(network.STA_IF)
w0.active(True)
e = espnow.ESPNow()
e.init()
peer = config.mac_of_receiver
e.add_peer(peer)
message = 'hello'
if e.send(peer, message, True) == True:
    print('ESPNow message received')
else:
    print('ESPNow message not received')

#receiver. Also connected to wifi
from esp import espnow
import config
import network

e = espnow.ESPNow()
e.init()
sta_if = network.WLAN(network.STA_IF)
if not sta_if.isconnected():
    sta_if.active(True)
    sta_if.connect(config.SSID, config.Wifi_password)
    while not sta_if.isconnected():
        pass
sta_if.config(ps_mode=network.WIFI_PS_NONE)
e.add_peer(sender_node_mac)
while True:
    host, msg = e.recv()
    if msg:
        print(msg)
        print(host)
```

closed 2021-07-13 by philwilkinson40