glenn20/micropython-espnow-images #8: ESP-NOW RSSI

davefes, 2021-11-03:
Would it be possible to add w0.status('rssi') to ESP-Now?  
Thanks,
Dave

glenn20, 2021-11-03:
A reasonable request Dave, but sadly not one I can deliver on.

The w0.config('rssi') is only calculated when connected to a wifi Access Point and is not provided through the ESPNow interfaces.

When I want to get the signal strength, I send a message to remote devices asking them to turn on their Access Points and then execute a scan on the local node. The scan results include the RSSI values. This can be useful for getting a static (or periodic) signal strength map for a collection of nodes. 

However, it may generate tto much drain on battery powered devices if you keep the AP on for too long or scan too frequently. (Alternatively, you can turn on the AP on a central, AC powered node and ask remote devices to execute a scan and report back the RSSI values.).

Not as straight forward as just calling w0.status('rssi') ;).

davefes, 2021-11-03:
As is it really only for antenna and QOS testing I can mains power
both ends.  Thanks for a procedure.

glenn20, 2021-11-21:
Well - I was premature in my earlier comment about lack of support for RSSI. As espnow uses an acton frame, the RSSI is available - and it can be accessed using the secret sauce to be found in the new Espressif espnow higher-level code: espnow.c:espnow_recv_cb() at https://github.com/espressif/esp-now/.

I am currently doing final debugging on a new feature that tracks the rssi of peers. The new feature:
- Adds the rssi data to the espnow packet format.
- Maintains an internal table (dict) of known peers and latest rssi readings: 
      {peer_mac1: (rssi, time_sec), peer_mac2: (rssi, time_sec), ...}
- You can get a reference to the dict using ESPNow.peers()
- So, you can monitor the rssi values (0 to -100dBm) for all peers by monitoring this dict.
- Using the internal peer table of mac addresses has a few other subtle but useful benefits too.

I think this will be really useful for managing IoT espnow networks of devices. It still needs a bit of work - but progressing nicely.

If you have any questions or suggestions about the implementation let me know.

Cheers, Glenn.

davefes, 2021-11-21:
Good news!  Thanks for persisting with this challenge.

glenn20, 2022-04-18:
The latest images and PR6515 now include support for RSSI so I'll mark this closed. Thanks for prompting me to investigate this further.

closed 2022-04-18 by glenn20