glenn20/micropython-espnow-images #9: HTTPS error

mdaeron, 2021-11-12:
I'm using a Lolin S2 mini board flashed with `firmware-esp32s2-GENERIC_S2.bin` from `20211111_espnow-g20-v1.17-142-g8e1fcd490`. The following code

````py
import network, time
from urequests import get
sta_if = network.WLAN(network.STA_IF)
sta_if.active(True)
sta_if.connect(ssid, wifipw)
time.sleep(3)
if sta_if.isconnected():
    get('https://example.com')
````
 yields the following error:

````
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "urequests.py", line 116, in get
  File "urequests.py", line 62, in request
OSError: [Errno 12] ENOMEM
````

That error does not occur if I use http instead of https, nor if I use the stock firmware. Is this issue out of scope for this fork ? I don't know if there is an "official" formware I can build (without ESP-NOW) and test for https behavior.

glenn20, 2021-11-12:
The memory requirements for https are challenging for micropythn boards and you are hitting a memory limit on the GENERIC_S2 target (which does not support PSRAM). The Lolin S2 Mini does have PSRAM so the easiest solution would be to use an image which supports the PSRAM on the S2 Mini.

The micropython daily builds include an image for the S2 Mini (https://micropython.org/download/LOLIN_S2_MINI/). Be aware these are builds from the daily main branch - not a release.

The number of supported S2 targets are rapidly multiplying in recent days in micropython main. I'll add those new targets to my espnow builds as well.

mdaeron, 2021-11-12:
Excellent, thanks.

closed 2021-11-12 by mdaeron

glenn20, 2021-11-13:
I've just uploaded new builds for all the supported esp32 target boards (excluding the GENERIC_D2WD target due to image size limitations). This includes firmware-esp32-LOLIN_S2_MINI.bin - which I have NOT tested, but I presume should incldue support for the onboard PSRAM which should alleviate the memory problems with ssl.

Note - these images are also built with the max_heap_limit from PR#7963 which is relevant to the GENERIC_S2 target (ie. S2 targets without PSRAM).

mdaeron, 2021-11-13:
Just ran the code above with your new Lolin S2 Mini firmware, and it works fine. I'm keeping this firmware on this board for now, please let me know if you'd like me to run other tests.