glenn20/micropython-espnow-images #11: ESP8266 and MQTT issue

davefes, 2021-12-18:
Posting for someone else at [https://forum.micropython.org/viewtopic.php?f=16&t=11635&p=63405#p63405](url)

Again I did the following:
>>> import upip #No Problem
>>>upip install('micropython-umqtt.simple')
Resulted in a syntax error.

Should have responded with:
Installing to: /lib/
Warning: micropython.org SSL certificate is not validated
Installing micropython-umqtt.simple 1.3.4 from https://micropython.org/pi/umqtt.simple ... 3.4.tar.gz

NOTE: The ESP-01S was connected to my WiFi network.

He seems to be using the latest 1M GENERIC ESP8266 image.

Thanks,
Dave

mmliam, 2021-12-18:
I'm the one for which davefes posted the issue #11; just created an account.
Here's some additional information:
This is the version that I don't have any issues with; it installs MQTT into the ESP-01S LIB
esp8266-1m-20210902-v1.17

This is the version that failed:
20210903_espnow-g20-v1.17
Actual file name: firmware-esp8266-GENERIC_1M (581KB)

BTW: There is another slightly larger bin-file in the same folder that I haven't tried:
firmware-esp8266-GENERIC (615KB)

mmliam, 2021-12-19:
FOUND THE PROBLEM:
Dumb programming:
Issued upip install.... instead of upip.install...

davefes, 2021-12-19:
Not quite there:

With build [firmware-esp8266-GENERIC_1M (581KB)] re-installed,
I am now going to try specifying a fixed channel to see if that speeds re-connection.

UPDATE:
Tried issuing the fixed channel assignment [sta_if.config(channel=11)]

Resulted in the same error as the standard build:
File "main.py", line 18, in <module>
OSError: AP required

MicroPython v1.17-12-g26e539a1b on 2021-09-03; ESP module (1M) with ESP8266
Type "help()" for more information.
>>>

mmliam, 2022-01-02:
re: [https://micropython-glenn20.readthedocs.io/en/latest/library/espnow.html#espnow-and-wifi-operation](url)

EXCERPT from EXAMPLE 2:
`e = espnow.ESPNow()
e.init()

w0 = network.WLAN(network.STA_IF)
w0.active(True)                        # Set channel will fail unless Active
w0.config(channel=6)
w0.connect('myssid', 'myppassword')
`
This code excerpt suggests the with ESPNow it should be possible to set a fixed WiFi channel for the Station-mode; as stated in previous post this throws the error "OSError: AP required".
Is this an error in ESPNow or the referenced micropython webpage?

glenn20, 2022-04-18:
As noted in Issue #12, the esp8266 api does not permit setting the channel of the STA_IF interface on the ESP8266.

However, you can set it by setting the channel on the AP_IF interface an then turning off the AP_IF. The STA_IF and AP_IF always run on the same channel:
```
import network

w0, w1 = (network.WLAN(i) for i in (network.STA_IF, network.AP_IF))
w0.active(True)
w0.disconnect() # Because esp8266 automatically reconnects to last AP
w1.active(True) # This turns on the Access Point
w1.config(channel=6) # STA and AP lways use the same channel
w1.active(False) # Turn off the Access Point
```

closed 2022-04-18 by glenn20