glenn20/micropython-espnow-images #12: WiFi Channel setting in Station-Mode not functioning as per glenn

mmliam, 2022-01-11:
re: Support for the ESP-NOW protocol
 https://micropython-glenn20.readthedocs.io/en/latest/library/espnow.html#espnow-and-wifi-operation

Excerpt:

Example 2: Send and receive ESPNow traffic on AP_IF interface:

`import network
from esp import espnow

peer = b'feedee'                       # MAC address of peer
e = espnow.ESPNow()
e.init()

w0 = network.WLAN(network.STA_IF)
w0.active(True)                        # Set channel will fail unless Active
w0.config(channel=6)
w0.connect('myssid', 'myppassword')

w1 = network.WLAN(network.AP_IF)
w1.config(hidden=True)                 # AP_IF operates on same channel as STA_IF
w1.active(True)

e.add_peer(peer, None, None, network.AP_IF)  # Register peer on AP_IF
e.send(peer, b'ping')                  # Message will be from AP_IF mac address

print('Send me messages at:', w1.config('mac'))

`

NOTE: w0.config(channel=6)

When this code is tried in an ESP8266 the follow error occurs:
"OSError: AP required"
Yet the above code clearly indicates that w0 is a station object [w0 = network.WLAN(network.STA_IF)].

In my code I'm only declaring a station object; do I need to declare access-point object also (as in the code above), even if the device doesn't need to be an access point just to fix the Wifi channel for station object?

glenn20, 2022-01-24:
Standard micropython images don't allow setting the channel on STA_IF interfaces for esp32 or esp8266 (because the Espressif SDKs didn't allow it). The espnow branch patches the networking code to allow setting the channel on STA_IF (for esp32 only!!) because this is now supported by recent versions of the Espressif ESP32 SDK - so this is not supported on esp8266!!

I BELIEVE changing the channel on the STA_IF on esp8266 is still NOT supported by the Espressif 8266 sdk. However, I confess I have not tested this belief recently. I will put it on my list of things to check.

In the meantime, I suggest you use the AP_IF interface instead if you want to be able to set the channel.

mmliam, 2022-01-25:
Hi Glenn,
Thanks for the reply. Correct me if I'm wrong, I didn't think the AP mode provided the same functionality as the Station-mode. The device I'm planning will awaken from deep-sleep (to save power) on a sensor activation, transmit the sensor status then go back to deep-sleep, awaiting another sensor activation. As I was using the station-mode, I wanted to fix the channel, to save the channel scanning time the unit takes to reconnect after coming out of deep-sleep (rapid re-connection is required for this application).

Would I be able to accomplish this same functionality using AP-mode?

glenn20, 2022-04-06:
Hi MIke,
(Answer may be redundant by now).
I believe that you can send an espnow message from the AP interface as you are proposing. I know it can be done with the esp32, but I'd need to check to confirm it also works with the 8266 (but I believe it does). You can check for yourself if you want to - but I will get to confirming this in the next few weeks.

dtuando, 2022-04-10:
You do not need to set channel if sending on STA_IF... Only on AP_IF

glenn20, 2022-04-18:
I have that the Espressif esp8266 API provides no support for setting the channel on the STA_IF interface. 

However, you can use the following snippet fo code to set the channel on the STA_IF:

`
import network

w0, w1 = (network.WLAN(i) for i in (network.STA_IF, network.AP_IF))
w0.active(True)
w0.disconnect()   # Because esp8266 automatically reconnects to last AP
w1.active(True)   # This turns on the Access Point
w1.config(channel=6)    # STA and AP lways use the same channel
w1.active(False)    # Turn off the Access Point
`
ie. activate both interfaces, ste the channel on the AP_IF (AP_IF and STA_IF alwys use the same channel, then turn off the AP_IF. Thsi will leave the STA_I running on channel 6. I have confrmed this on my own esp8266 devices.

I will close this issue now. Feel free to re-open if you continue to have issues.

closed 2022-04-18 by glenn20