glenn20/micropython-espnow-images #13: i2c / readfrom_mem cause core 0 panic

Sandroggy, 2022-02-12:
Hi !

Can't get firmware-esp32-GENERIC.bin to work with i2c, the error comes from readfrom_mem()

Might be related to this : https://forum.micropython.org/viewtopic.php?t=11330

glenn20, 2022-04-06:
Thanks for the bug report. As confirmed in the forum post referenced above, this is a known bug with esp idf 4.4. I'll make a future build against the same build environment as v1.18 which should adress this issue.

luiz-brandao, 2022-04-15:
Thanks @glenn20! In the mean time, is there a guide on how to compile and image for ESP32?

glenn20, 2022-04-16:
Sure. I just built my build environments following the instructions at https://github.com/micropython/micropython/blob/master/ports/esp32/README.md

To support all the newer esp32 variants I need to build with IDF v4.4. I'll build a new esp32-GENERIC image with IDF v4.2.

I'll also try updating to a newer v4.4 release as it may have the bugfix for i2c by now.

glenn20, 2022-04-18:
I have put two new images up at: https://github.com/glenn20/micropython-espnow-images/tree/main/misc - one of which may fic the i2c issue. @luiz-brandao or @Sandroggy : if you have the interest, time and inclination, could you please check both of these images and let me know if either or both of these fix the i2c problem. One is built with the v4.0.2 ESP IDF and the other is with an updated v4.4 ESP IDF.

glenn20, 2022-04-18:
Actually, I have just checked and I can see that the v4.4 image I originally uploaded will still have the problem. I have just updloaded a new v4.4 image which includes the relevant bugfix. Please test this newest image.

**The details:**
The latest v4.4 IDF does include the i2c bugfix - and micropython master works fine with that latest version. However micropython release v1.18 compiled against the latest update to v4.4 IDF does not work (no repl prompt). So, micropython v1.18 is built against an earlier version of v4.4 which does not include the fix. I have applied the i2c fix against this earlier version of v4.4 and rebuilt the image. Hopefully it works this time. I will build all my future v1.18 images against this patched IDF, so please let me know if it solves the i2c bug.

luiz-brandao, 2022-04-18:
Hey @glenn20, I tried the image [firmware-esp32-GENERIC-idf44-v1.18.bin](https://github.com/glenn20/micropython-espnow-images/blob/main/misc/firmware-esp32-GENERIC-idf44-v1.18.bin) and from my limited testing, it seems to have solved the I2C bug. I was getting the error as soon as I tried to instantiated the machine.I2C class, but not anymore. All peripherals seem to be properly initialized. Awesome work! Thank you!

glenn20, 2022-04-18:
Excellent. Thanks for testing. I'll use those patches for any new v1.18 images I build.

closed 2022-04-18 by glenn20

luiz-brandao, 2022-04-19:
@glenn20 not sure if directly related but I started having memory issues when trying to use the new `aioespnow` module. I created a ticket at: https://github.com/glenn20/micropython/issues/1