glenn20/micropython-espnow-images #15: Glenn20 Post on AP & STA channel sharing

mmliam, 2022-02-27:
Hi Glenn20

In your post on WiFi: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv419esp_wifi_set_config16wifi_interface_tP13wifi_config_t

Note 3 reads:
Attention

    3. ESP32 is limited to only one channel, so when in the soft-AP+station mode, the soft-AP will adjust its channel automatically to be the same as the channel of the ESP32 station.
 
Unfortunately, the microPython build for the ESP8266 does not allow a station object to set a fixed channel, it scans till it find the channel for the target SSID. This results in a variable & longer than desirable connection time.

Presuming the ESP8266 is also limited to one channel.  Would the reverse of your note 3 above allow the station mode to be fixed. In otherwords if an AP object is created first and assigned a channel [e.g. netObjAP.config(channel=11)], then would a station mode object created next [e.g. netObjSTA.connect(SSID, PASS)] be forced to use the same channel?

Mike

glenn20, 2022-04-06:
I believe that may well be true. If you set the channel using the AP interface, then the STA interface will use the same channel (until you override that by connecting to an external access point). However it is possible that initialising the STA interface will reset the channel (I ave not checked for mysef recently).

I suggest testing and seeing if it works. Alternatively, as suggested elsewehere, you can just use the AP interface to send and receive espnow messages.

glenn20, 2022-04-18:
As noted in Issue #12, you can indeed set the channel for the STA_IF on the esp8226 folllowing something like the recipe you suggested above:

`
import network

w0, w1 = (network.WLAN(i) for i in (network.STA_IF, network.AP_IF))
w0.active(True)
w0.disconnect() # Because esp8266 automatically reconnects to last AP
w1.active(True) # This turns on the Access Point
w1.config(channel=6) # STA and AP always use the same channel
w1.active(False) # Turn off the Access Point
`

If you have any problems feel free to re-open this issue.

closed 2022-04-18 by glenn20