glenn20/micropython-espnow-images #16: Can't flash 8266

wokibe, 2022-03-21:
Downloaded from [micropython-espnow-images](https://github.com/glenn20/micropython-espnow-images)/20211111_espnow-g20-v1.17-142-g8e1fcd490 the imagefile for 8266 and prepended 2011111_ to the name:
(esp) pi@raspberrypi:~/Projects/espnow $ ls -l 20211111_firmware-esp8266-GENERIC.bin 
-rw-r--r-- 1 pi pi 127797 Mar 21 08:49 20211111_firmware-esp8266-GENERIC.bin

An attempt to flash an 8266 failed:
(esp) pi@raspberrypi:~/Projects/espnow $ esptool.py --port /dev/ttyUSB0 --baud 115200 
	write_flash --flash_size=detect -fm qio 0 20211111_firmware-esp8266-GENERIC.bin 
esptool.py v3.0
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: a8:48:fa:cf:00:e8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings. <<<<<<<<<<<<<<<<<<<<!!!
Compressed 127800 bytes to 24585...
Wrote 127800 bytes (24585 compressed) at 0x00000000 in 3.0 seconds (effective 344.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
(esp) pi@raspberrypi:

Verified that the 8266 is OK with a standard image file

glenn20, 2022-04-06:
That image should be 614592 bytes in size. It appears your download was incomplete. Please try the download again.

wokibe, 2022-04-06:
Thanks for your help. I was using a RaspberryPi and the by default
installed browser is Chromium did not download the binary firmware
correctly.
I installed firefox-esr and now the firmware with esp-now is installable!
Please close the issue.
best regards,
Wolfgang Kittenberger

glenn20, 2022-04-18:
Great - glad you got it working.

closed 2022-04-18 by glenn20