glenn20/micropython-espnow-images #18: ESP-NOW and WIFI Same time?

dtuando, 2022-04-09:
I'm trying to follow the docs to get WiFi and ESP-NOW simultaneously. 

`import network
import machine
import ubinascii
from esp import espnow
import mrequests as requests
from time import sleep

import gc
gc.collect()

try:
  import usocket as socket
except:
  import socket

# To print this device's MAC address:
sta_if = network.WLAN(network.STA_IF)
ap_if = network.WLAN(network.AP_IF)

###Get Device MAC###
sta_mac = ubinascii.hexlify(network.WLAN(network.STA_IF).config('mac'),':').decode('utf-8')
ap_mac = ubinascii.hexlify(network.WLAN(network.AP_IF).config('mac'),':').decode('utf-8')

print (sta_mac)
print (ap_mac)

### Initiate AP_IF for ESP-NOW ###
sta_if.active(False)
ap_if.active(True)
ap_if.config(channel=7, hidden='True')

peer = b'\x08\x3a\xf2\x51\x68\x29'
e = espnow.ESPNow()
e.init()
e.add_peer(peer, ifidx = network.AP_IF)

### STA_IF Connection ###

essid = 'SSID-Redacted'
passw = 'PASSWORD-Redacted'

def do_connect():
    ap_if.active(False)
    sta_if.active(True)
    if not sta_if.isconnected():
        print('connecting to network...')
        sta_if.connect(essid, passw)
        while not sta_if.isconnected():
            pass
    print('network config:', sta_if.ifconfig())

t = ""
h = ""

print('waiting for message...')
while True:
    host, msg = e.irecv()     # Available on ESP32 and ESP8266
    m = msg.decode('utf-8')
    temp = "Temp"
    hum = "Hum"
    if msg:             # msg == None if timeout in irecv()
        if temp in m:
            t = m
            print(t)
        if hum in m:
            h = m
            print(h)
            do_connect()
            sleep(2)
            
`

The only way I was able to get the STA_IF to connect to my router was to Disable ap_if first... what am i doing wrong?
Is it possible to have both WIFI and ESP-NOW running together?

glenn20, 2022-04-10:
As many others have commented previously there are some mysteries to the way that wifi works on the ESP32 and it is not helped by the sometimes ambiguous espressif documentation.

I can confirm that I am successfully using wifi and espnow together - but perhaps not using exactly the same procedure you are. In fact, the kind of behaviour you are seeing seems to be about the quirks of using wifi on the esp32 (not just espnow).

I am going to speculate that the espressif wireless code doesn't let you connect to a wifi access point with the STA interface while the AP interface is active. (Or that you can't connect to an access point that is operating on a different channel from the AP interface). I can't test this because I am away from home at the moment.

I recommend: 

- connect to the access point with your STA interface first (as in Example 2 of the docs)
- then activate the AP interface 
  - it will operate on the same channel as your access point - remember that there is only one real radio on the esp32/8266.

Otherwise, you can use the Example 1 methodology so you can avoid using the AP interface altogether.

Good luck - and let me know if these recommendations don't work.

dtuando, 2022-04-10:
thank you for the quick response!
I will attempt to get the STA connected first, Then i will activate AP and come back with my results.

Thank you for the great project!
How can I buy you cup of coffee or a beer?

dtuando, 2022-04-10:
So connecting to my access point through STA_IF first, and AP_IF second brings about an immediate error

> OSError: Wifi Invalid Mode

Attempting first Method

dtuando, 2022-04-10:
OK! So I was able to get it to have both 

Here is the code i was able to get it running off of.

```
import network
from time import sleep

sta_if = network.WLAN(network.STA_IF)
sta_if.active(True)

ap_if = network.WLAN(network.AP_IF)
ap_if.active(True)

sta_passw = 'CHANGE ME'
sta_essid = 'CHANGE ME'

def ap_sta():
    if ap_if.active() == True:
        if sta_if.active() == True:
            sta_if.connect('sta_essid', 'sta_passw')
            ap_if.config(essid='AP_ESSID_CHANGEME')
            sleep(2)
            print('AP SSID Updated!')

while True:
    if sta_if.isconnected():
        print('STA ifconfig:', sta_if.ifconfig())
        print('AP_ifconfig:', ap_if.ifconfig())
        break
    else:
        sleep(5)
        ap_sta()
        pass
```

without the sleep timer i think it causes errors somewhere. I'm not quite sure how to debug where its happening at. If you could take a look at this code Please let me know if it helps anyone with similar issues.

dtuando, 2022-04-10:
if you load this into the boot.py, you may run esp now like normal in main.py and can even set the channels to match the sender board if using AP_IF on sender as well!

just add ``` ap_if.config(channel=CHANNEL_NUMBER) ``` after the ``` ap_if.config(ap_essid) ``` statement.

glenn20, 2022-04-18:
Hmmm - I think you can simplify the code so the sleep(5) *should* be unecessary. I believe that it can take a little while after calling sta_if.connect() before isconnected() becomes true.

Try the following (warning untested code snippet):

```
sta_if.active(True)
sta_if.disconnect()  # Stop esp8266 from automatically reconnecting to last AP
while sta_if.isconnected():
    time.sleep(0.2)

sta_if.connect(sta_essid, sta_passwd)
while not sta_if.isconnected():
    time.sleep(0.2)

ap_if.active(True)
ap_if.config(essid=ap_essid)

print('STA ifconfig:', sta_if.ifconfig())
print('AP_ifconfig:', ap_if.ifconfig())
print('Channel:', ap_if.config('channel'))
```

closed 2022-04-18 by glenn20