glenn20/micropython-espnow-images #20: LR-mode

davefes, 2022-04-30:
There is an open issue here:[https://github.com/micropython/micropython/issues/8510](url)
My tests indicate that normal WiFi and ESPNow work down to the same RSSI, ie -93dBm.  Appears that ESPNow doesn't reduce the receiver bandwidth to improve signal to noise ratio.

glenn20, 2022-05-04:
Thanks for the report. The LR mode is not supported in the micropython releases, but it is supported in the espnow releases. The espnow patches add support for:

```
w0 = network.WLAN(network.STA_IF)
w0.config(protocol=network.MODE_LR)
```

Documentation for LR mode is available at: 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-protocol-mode

davefes, 2022-05-04:
I didn't make it clear that I was asking if you could put LR-mode into your ESPNow images.  I was not aware that LR-mode was already included.  I tried your code snippet before making w0.active(True) but get this error message:
Error initialising espnow: Wifi Unknown Error 0x0102

davefes, 2022-05-04:
Found a posting about this error and tried making the local (receiver) AP_IF.  That error disappears.  However, when I do that at the remote (sender) end I now get:
Error during execution: (-12396, 'ESP_ERR_ESPNOW_IF')

glenn20, 2022-05-04:
> I didn't make it clear that I was asking if you could put LR-mode into your ESPNow images. I was not aware that LR-mode was already included. I tried your code snippet before making w0.active(True) but get this error message: Error initialising espnow: Wifi Unknown Error 0x0102

I just ran some checks and it seems that it works fine so long as the STA_IF has been made active(True) at least once since the last hard reset or power cycle.

So, reverse my previous advice to set the protocol before calling active(True):
```
w0 = network.WLAN(network.STA_IF)
w0.active(True)
w0.config(protocol=network.MODE_LR)
```

Note that it does not throw the error even if w0 is not active - just so long as it has been set active at least once since the last reboot.

glenn20, 2022-05-04:
> Found a posting about this error and tried making the local (receiver) AP_IF. That error disappears. However, when I do that at the remote (sender) end I now get: Error during execution: (-12396, 'ESP_ERR_ESPNOW_IF')
> 
> I realised that when you put an end into AP-mode then you have to change the mac, however all combinations of AP/STA and mac address still give the above error.

Remember that when you call add_peer() on the remote to register the local end, you should set the ifidx parameter to 1 (network.AP_IF), so that the espnow stack knows to send messages to that peer through the AP_IF interface. eg:
```
w1 = network.WLAN(network.AP_IF)
w1.active(True)
w1.add_peer(peer_mac, ifidx=network.AP_IF)
```

You can check with `e.get_peers()` which should return a "1" as the second last member of the tuple for your host.

davefes, 2022-05-04:
I will post code on a new thread on the Micropython forum.

Thanks for your help

davefes, 2022-08-06:
Using MicroPython v1.19.1-espnow-6-g44f65965b I am having another go at LR-mode.

In your code snippet above I understand the need to have ifidx=network.AP_IF to talk to a local which is set up for AP.  The snippet suggests that the remote is also on AP.  Should both units be on AP?

How is longer range observed?  I would expect the RSSI reading to be different between the modes at the same distance or be able to move the devices further apart before the link drops out.  I see both modes drop out at -95dBm at the same distance and re-establish at -93dBm.  I have seen comments about -105dBm (LR-mode), in which case I would expect to see a change in this go/no-go threshold at the receive end. 

I have tried looking around for the source code, do you have a link?

glenn20, 2022-08-07:
I believe my snippet was based on that you were already using the AP_IF. If all your devices are ESP32s, you shouldn't need to use the AP_IF at all on any of your devices as you can send and receive ESPNow messages just fine with the STA_IF (and that is simpler).

I have not tested LR_MODE to see if it provides improved range. I have only tested that connections work. A user pointed me at the following video [6:05 on video](https://youtu.be/oz0a7Ur7nko?t=365) which suggests success. To replicate the conditions they used in that test (and using my wifi.py module from <https://github.com/glenn20/micropython-espnow-utils>):

```python
import wifi, network, espnow

sta, ap = wifi.reset(protocol=network.MODE_LR)  # STA on disconnected, AP off, force LR_MODE
wifi.status()  # print details to confirm settings

peer=b'xxxxxx'
enow = espnow.ESPNow()
enow.active(True)
enow.add_peer(peer)
....
```
That should initialise everything to the same state as they used in that test. Note that your devices will not be able to connect to a wifi network in this mode (11B,11G,11N disabled).

I'm not sure how to make LR_MODE work with wif mode - it might take some experimentation.

davefes, 2022-08-07:
Actually, all my testing so far has been with both units on STA mode.  I can confirm that the connection works, just not appear to work better.  I do not want to use WiFi on these test units.  That can wait for another "day". 

Thank you for the link to the video and the espnow-utils.

closed 2022-08-10 by davefes