glenn20/micropython-espnow-images #22: Continuous crash on boot - Guru Meditation Error: Core  1 panic'ed

mertozal, 2022-07-09:
Hello everyone,

Trying to flash "20220709_espnow-g20-v1.19.1-espnow-6-g44f65965b/firmware-esp32-LILYGO_TTGO_LORA32.bin" but after flashing successfully, ESP32 continuously resets. But if i erase and flash original esp32 img it works. 

Error:

`Performing initial setup
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400934da  PS      : 0x00060030  A0      : 0x8008f724  A1      : 0x3ffce200  
A2      : 0x2ffc0794  A3      : 0x3ffe7660  A4      : 0x3f402fec  A5      : 0x3f403028  
A6      : 0x00000001  A7      : 0x00000002  A8      : 0x800934f0  A9      : 0x3ffce1e0  
A10     : 0x15122500  A11     : 0x00039660  A12     : 0x00000001  A13     : 0x00000020  
A14     : 0x3ffe7660  A15     : 0x0000000c  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x15122500  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace:0x400934d7:0x3ffce2000x4008f721:0x3ffce220 0x4014f88d:0x3ffce260 0x40112bc8:0x3ffce290 0x400deaa3:0x3ffce2c0 0x400e54fd:0x3ffce2f0 0x400e562d:0x3ffce310 0x400efc45:0x3ffce330 0x400f051d:0x3ffce360 0x400f32a1:0x3ffce380 0x400f407a:0x3ffce3c0 0x400f6125:0x3ffce3f0 0x400f0618:0x3ffce480 0x400dea95:0x3ffce5d0 0x400e54fd:0x3ffce600 0x400e562d:0x3ffce620 0x40084b7d:0x3ffce640 0x400dea36:0x3ffce6e0 0x400e54fd:0x3ffce750 0x400e562d:0x3ffce770 0x40084b7d:0x3ffce790 0x400dea36:0x3ffce830 0x400e54fd:0x3ffce8a0 0x400e5526:0x3ffce8c0 0x400f2b16:0x3ffce8e0 0x400f2e7a:0x3ffce970 0x400d6f0f:0x3ffce9a0 

ELF file SHA256: 121133f62ba9dd9a

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4396
ho 0 tail 12 room 4
load:0x40078000,len:13884
load:0x40080400,len:3340
entry 0x40080618`

glenn20, 2022-07-10:
Thanks for the bug report. Unfortunately I don't have a similar device to test. This may be an issue with the correct use of the IDF to use in the build. Unfortunately micropython don't provide any information on the build environment used for the official images, so I have to make some guesses. The images I have uploaded work on all the devices I have, including my esp32 devices.

I have uploaded two new images for the LILYGO_TTGO_LORA32 to <https://github.com/glenn20/micropython-espnow-images/tree/main/misc>. Compiled against the v4.0.2 and v4.2 IDF. One of those may work for you. My images are built against the v4.4 IDF.

Please let me know if either of these work for you. That'll help me with my attempts to deduce the optimum build environments for the different chip flavours.

mertozal, 2022-07-10:
Thank you for your fast response and solution. It works like a charm now! 

I tested both images and i can confirm that both of them working with TTGO LORA32(T3_V1.6.1) board.

Best,

closed 2022-07-10 by mertozal

glenn20, 2022-07-10:
Excellent. Thanks for the feedback that helps me refine my build environments...