glenn20/micropython-espnow-images #23: Errors on latest 19.1 for ESP32 Generic

davefes, 2022-07-13:
Latest 19.1 firmware-esp32-GENERIC.bin

```
e.add_peer(peer, None, None, network.AP_IF) #  register peer on AP_IF
TypeError: can't convert NoneType to int
```

Had to do:
`e.add_peer(peer, ifidx=network.AP_IF) #  register peer on AP_IF`

```
w1.config(hidden=True)              # AP_IF operates on same channel as STA_
OSError: Wifi Invalid Mode
```

I assume this is quite important to ensure the relevant interfaces are on the
right channel.

glenn20, 2022-07-17:
Thanks for the report Dave. I hadn't noticed this user facing change creep in when I streamlined the argument processing code in ESPNow.config() method - although that change was quite some time ago. I will investigate a fix...

**Note:** This call won't change the channel. It tells the ESP-NOW software stack to send messages to this peer through the AP_IF interface, instead of the STA_IF (which is useful if you have both interfaces active - otherwise it sends from whichever interface is turned on). It also means that the peer will see the message coming from the mac address of tthe AP_IF instead of the STA_IF.

davefes, 2022-07-17:
I should have mentioned that I was trying to use Example 2 when I ran into those errors.

Your Note does not apply to Example 1 in the docs, right?

glenn20, 2022-07-17:
That's right - my note was referring to the `e.add_peer(peer, None, None, network.AP_IF)` or `e.add_peer(peer, ifidx=network.AP_IF)` calls.

closed 2022-07-19 by davefes