glenn20/micropython-espnow-images #25: ifconfig behaviour on 1.19.1

davefes, 2022-07-21:
```
    w0.active(True)

    print('connecting to hotspot...')

    utime.sleep(1) #  was getting Guru Meditation Error without this delay
                            #  when doing w0.config(), ie a static IP

    w0.ifconfig((config.WiFi_device, '255.255.255.0', config.gateway, '8.8.8.8'))

    w0.connect(config.hotspot, config.password)
```
Haven't seen this issue on mainline.

glenn20, 2022-07-30:
Hmm - I tried this sequence on my esp32 devices (using v1.19.1-6.g44f65965b) and it works fine... with or without the sleep().

I've checked after a hard or soft reset and seems fine (repeated 43  times).

If I put this sequence into my main.py it runs fine for a hard reset (and also for the first soft reset after a hard reset), however for the second soft reset after a hard reset, I get:
```
MPY: soft reboot
connecting to hotspot...
E (36703) wifi:sta is connecting, return error
Traceback (most recent call last):
  File "main.py", line 6, in <module>
OSError: Wifi Internal Error
```
However, it is connected to the wifi just fine.

This not toyally unexpected - remember that micropython does NOT reset the peripherals (including wifi) after a soft reset.

Can you provide more details on the device and confirm which image you are using?

Have you run any code which touches the wifi peripherals before this code fragment?

Does, this only happen immediately after a soft or hard reset - or at any time after the device has booted?

davefes, 2022-07-30:
MicroPython v1.19.1-espnow-6-g44f65965b on 2022-07-09; ESP32 module with ESP32
ESP32 DevKitC WROOM32U with rev 1 chip

main.py (first time there is any WiFi activity

```
import utime
from getUTC import getntptime
import gc
import ramfs
import uos
from watchdog_timer import wdt_callback
from watchdog_timer import wdt_feed
from wifi_functions import connect_w0
from wifi_functions import disconnect_w0
from mail_functions import send_csv
import config

#  Software WDT set for 70 seconds
wdt_callback() #  set up for 70 seconds

wdt_feed()
connect_w0()
wdt_feed()
```
connect_w0 in wifi_functions.py
```
import network
import utime
import machine
import config

w0 = network.WLAN(network.STA_IF)     # create station interface
ap_if = network.WLAN(network.AP_IF)   # create access point interface

def connect_w0():
    count = 0

    if ap_if.isconnected():
        ap_if.disconnect()
        print ('AP started in the connected state, but now is disconnected')
    else:
        print ('AP started in the disconnected state')

    utime.sleep(1)

    ap_if.active(False) #  ensure AP is inactive

    if w0.isconnected():
        w0.disconnect()
        print ('w0 started in the connected state, but now is disconnected')
    else:
        print ('w0 started in the disconnected state')

    utime.sleep(1)

 #  both interfaces should now be disconnected

 #  now make w0 active and connect
    w0.active(True)

    print('connecting to hotspot...')

    utime.sleep(1) #  was getting Guru Meditation Error without this delay
                   #  and when doing w0.config(), ie a static IP

    w0.ifconfig((config.WiFi_device, '255.255.255.0', config.gateway, '8.8.8.8'>

    try:
        w0.connect(config.hotspot, config.password)
    except OSError as error:
        try:
            with open('errors.txt', 'a') as outfile:
                outfile.write(error + '\n')
        except OSError:
            pass
```
Soft reset OK first time
Hard reset: fails
then Soft reset fails every time after that until I put that line back in.

```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4009839e  PS      : 0x00060933  A0      : 0x80095c21  A1      : 0x3ffd0050  
A2      : 0x0000004c  A3      : 0xffffffff  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x0000cdcd  A9      : 0x0000abab  
A10     : 0x00060920  A11     : 0x0000cdcd  A12     : 0xb33fffff  A13     : 0x00060923  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000001  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000004c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace:0x4009839b:0x3ffd00500x40095c1e:0x3ffd0070 0x4015a90e:0x3ffd00b0 0x4015a961:0x3ffd00d0 0x4015b85e:0x3ffd00f0 0x4015a2ab:0x3ffd0120 0x4015a2ca:0x3ffd0150 0x401e6461:0x3ffd0170 0x401e6524:0x3ffd01d0 

ELF file SHA256: ebc3e7139c1cd63d
```
I get rid of that internal WiFi error by making sure I am disconnected rather then doing a hard reset every time.  Maybe, I have created another problem?

glenn20, 2022-07-30:
Thanks. And if you change main.py to:

```
from wifi_functions import connect_w0
connect_w0()
```

do you get the same errors? 

Or just change main.py to:

```
import network
w0=network.WLAN(0)
w0.active(True)
print('connecting to hotspot...')
w0.ifconfig(('XXX.XXX.XXX.XXX', '255.255.255.0', 'XXX.XXX.XXX.XXX', '8.8.8.8'))
w0.connect('XXXXXX, 'XXXXXXXXX')
```

(Change all the XXXs accordingly).

davefes, 2022-07-30:
```
import network
import utime
import config

w0 = network.WLAN(network.STA_IF)     # create station interface

if w0.isconnected():
    w0.disconnect()
    print ('w0 started in the connected state, but now is disconnected')
else:
    print ('w0 started in the disconnected state')

utime.sleep(1) # needs this or the one before ifconfig

w0.active(True)

print('connecting to hotspot...')

#utime.sleep(1) #  was getting Guru Meditation Error without this delay
               #  and when doing w0.config(), ie a static IP

w0.ifconfig((config.WiFi_device, '255.255.255.0', config.gateway, '8.8.8.8'))
w0.connect(config.hotspot, config.password)

print ('waiting for 5 seconds so you can do a CTRL-C')
utime.sleep(5)
```
Seems the Guru Mediation Error often means that I have to re-flash the device.
A 1 second delay before w0.active(True) also works.  Leave them both out and do a
hard-reset.  Sometimes it takes two hard-resets to fail.

It appears doing a w0 disconnect() to get rid of the WiFi internal error without some delay
before w0.ifconfig() could be a problem.  I haven't seen this in mainline for the last few months of
doing it on my other systems.

glenn20, 2022-07-30:
> Seems the Guru Mediation Error often means that I have to re-flash the device. A 1 second delay before w0.active(True) also works. Leave them both out and do a hard-reset. Sometimes it takes two [...]

Hmm - it should never execute the w0.disconnect() call after a clean reboot as w0.isconnected() should always return False - since the wifi interfaces have not yet been set active(True).

How do you know it is executing the w0.disonnect() call on bootup?

davefes, 2022-07-30:
The problem first arose when doing soft-resets from rshell, but as above it looks like I can provoke it doing hard-resets.

> How do you know it is executing the w0.disonnect() call on bootup?

It appears that me doing a if w0 is connected then I do a w0 disconnect to get rid of WiFi internal error without ...
(the first 5 lines in the above script)

If I leave that test off and only do hard-resets then things are OK.    Maybe, it is bad programming practice but I often find that when dealing with WiFi-related stuff that a short delay, in certain places, gets rid of problems.  

Perhaps it would be better if I removed that test and if anything fails (internal WiFi error) have the device do a machine.reset().

davefes, 2022-07-30:
Removing the if w0 connected test and relying on a machine.reset() if WiFi doesn't fire-up correctly seems at better approach.

closed 2022-07-30 by davefes

glenn20, 2022-07-31:
Thanks Dave. I'm thinking of doing some work on the esp32 wifi driver to improve these situations. It seems that a lot of espnow user issues (and for myself too) are related to the failure of the wifi driver to detect and reset the wifi peripheral propoerly on restart. The micropython wifi driver does not call esp_wifi_deinit() on soft reset and does not check the state of the peripheral on startup.

I think I'm going to need to come up with some test cases that show the issue only for wifi as I don't feel confident a PR would be accepted just because it fixes issues for espnow users.

davefes, 2022-07-31:
Test case:

```
w0 = network.WLAN(network.STA_IF)     # create station interface

#  now make w0 active and connect
w0.active(True)

print('connecting to hotspot...')

w0.ifconfig((config.WiFi_device, '255.255.255.0', config.gateway, '8.8.8.8'))

try:
    w0.connect(config.hotspot, config.password)
except OSError as error:
    try:
        with open('errors.txt', 'a') as outfile:
            outfile.write(str(error) + '\n')
    except OSError:
        pass

print (you have 5 seconds to do a 'ctrl C')
utime.sleep(5)
```
In rshell load this go to REPL and import test_file, then CTRL-C, then CTRL-D, then CTRL_C then look in errors.txt  Shouldn't take long.