glenn20/micropython-espnow-images #26: ESPNow interface won't accept config parameters on the Generic-SPIRAM variant

davefes, 2022-07-23:
#  set up the ESPNow interface
e0 = espnow.ESPNow()
e0.active(True)
print (e0.config('mac'))

>>> 
MPY: soft reboot
Traceback (most recent call last):
  File "main.py", line 1, in <module>
  File "controller.py", line 62, in <module>
ValueError: unknown config param
MicroPython v1.19.1-espnow-6-g44f65965b on 2022-07-09; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 
Appears not to set up any config values.

glenn20, 2022-07-23:
Interesting. It shouldn't be soft rebooting, but 'mac' is not a parameter of ESPNow.config(). I think you meant WLAN.config('mac') ;).

davefes, 2022-07-23:
O, dear I have too many projects on the go!
I believe Ctrl-D in rshell  returns ESPNow, without a WiFi connection, properly.  If I have a WiFi interface active then a hard-reset is required.

closed 2022-07-23 by davefes

glenn20, 2022-07-24:
I know the feeling. Yes, sometimes a soft reset leaves the hardware in a bad state and a hard reset is needed.