glenn20/micropython-espnow-images #28: Time to take iterating over the ESPNow object

davefes, 2022-09-11:
```py
    for mac, msg in e0:
        if mac == config.house:
            data = msg.decode('utf-8')
            print (data)

        elif mac is None:
            print ('no peers found')

        else:
            print ('Recv from {}: "{}"'.format(mac, msg))

        break
```
Again an observation, not really a problem, it seems to take about 1 second.  Is there a lot of processing going on in the background?

glenn20, 2022-09-12:
Hmmm - there should not be an undue amount of processing going on. That does seem very slow though.

I just ran a test on an esp32 and was able to receive 1000 message in 25 seconds (I think the limit was the sender which was an esp8266). I do need to update my transfer speed tests though, so will double check later this week - but for now I can confirm 1000 messages (~20 bytes long) in 25 seconds. I did not print the messages though.

davefes, 2022-09-12:
Reading through your answer and while typing a reply I realised where the 1 second was coming from.  In the implementation that I copied the e0 setup I had the following:
```py
e0.config(timeout=1000) #  listen for 1 second for any ESPNow traffic
```
I will now see if I can dispense with this configuration parameter!

closed 2022-09-12 by davefes

glenn20, 2023-04-01:
> Hmmm - there should not be an undue amount of processing going on. That does seem very slow though.
>
> I just ran a test on an esp32 and was able to receive 1000 message in 25 seconds (I think the [...]

I just wanted to document here that the 25ms per read limit was almost certainly due to a regression in the send() code. Send()ing data was inadvertently limited by a 25ms busy wait to confirm the packet was received (when using `sync=True` - the default setting). This has been fixed in the espnow-g20 branch, but I havent built any new images recentky which fix this (I have been waiting for 1.20).