glenn20/micropython-espnow-images #29: ESP32 ESPNow generic urequests

davefes, 2022-09-24:
Is this some reason, other than image size that urequests is not part of this image?.

glenn20, 2022-09-28:
Ah... yes. I discovered that the default build process for micropython images is NOT used for the release images (they use a slightly different board definition which includes urequests, mqtt and some other library modules frozen into the image). Now that I know about it, I will fix that in my next round of image builds. In the meantime you have to install your own urequests.py (or better - urequests.mpy) onto the filesystem of the image.

Note that a recent commit to the master branch removes this difference, so the default build is the same as the release build.

closed 2022-09-28 by glenn20

davefes, 2022-09-28:
I also noticed yesterday that mqtt.simple and mqtt.robust weren't in there as well so reverted to the standard v1.19.1 for a project.
Thank you.