glenn20/micropython-espnow-images #32: ESPNow.get_peer return value

krishnak, 2022-10-01:
ESPNow.get_peer(mac) (ESP32 only)[¶](https://micropython-glenn20.readthedocs.io/en/latest/library/espnow.html#espnow.ESPNow.get_peer)
Return information on a registered peer.

Returns:
(mac, lmk, channel, ifidx, encrypt): a tuple of the “peer info” associated with the mac address.

The Channel value returned here is 0 which means current channel - can you return the actual channel number - or else it defeats the purpose of finding the actual channel number. I can ignore cross talk messages if I get the channel number in the message

krishnak, 2022-10-01:
The value returned is as below

```
Rssi:-34
(b'\x0c\xb8\x15\xf2\xdb=', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', 0, 1, False)

```

glenn20, 2022-10-01:
> The Channel value returned here is 0 which means current channel - can you return the actual channel number - or else it defeats the purpose of finding the actual channel number. I can ignore cross talk messages if I get the channel number in the message

The behaviour here follows the behaviour in [`esp_now_get_peer()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html#_CPPv416esp_now_get_peerPK7uint8_tP19esp_now_peer_info_t) and just returns the value you provided when you called `add_peer()`. It doesn't query the actual channel of the device (I believe there is no way to determine that).

BTW. I believe cross-talk is not an unexpected occurence in wifi networks, but the wifi protocols normally mitigate the impacts (eg. APs advertise their channel) - but I'm not an expert on wifi ;). I use a custom discovery protocol for devices to advertise their presence (including the channel in use).

closed 2022-10-01 by glenn20

krishnak, 2022-10-01:
Yes I ended up adding an extra parameter on the broadcast message which gives the channel value.I thought it would be more convenient if this information was available in the peer data, as it have everything else

Interestingly the cross talk is appearing consistently on the device has both STA+AP enabled. and the STA is connected to the internet via router and the AP is purely for ESP NOW . This device finds packets on both the adjacent channels say 30% of the time. However when it sends back an ESP NOW acknowledgement data on the incorrect channel - the ESP Now node which sent it is not receiving it (as expected)

glenn20, 2022-10-02:
> Interestingly the cross talk is appearing consistently on the device has both STA+AP enabled. and the STA is connected to the internet via router and the AP is purely for ESP NOW . This device finds [...]

It is possibe that you *may* be able to reduce this cross talk if your STA is connected to a wifi network configured for HT40 operation (40MHz bandwidth). If you reconfigure your wifi router to use 20MHz bandwidth for that ssid, your STA+AP device will configure it's radio to HT20 mode which *may* reduce/eliminate the cross-talk you are seeing.