glenn20/micropython-espnow-images #33: Useful addition to ESPNow/WiFi documentaion

davefes, 2022-10-01:
After you have setup ESPNow/WiFi units to use a specific channel check that your router is not set to automatic channel selection.

glenn20, 2022-10-02:
Thanks Dave = I really welcome docs suggestions. Could you clarify what you mean by this. My briain is a bit slow today :-).

davefes, 2022-10-02:
I wondered why my ESPNow/WiFi system stopped working, I reasonably quickly ascertained that the router I was using had decided to change from channel 6 to channel 1.  I think the introduction of a another router on channel 11 BUT with 40MHz bandwidth was spilling-over onto channel 6.  I read somewhere that adjacent channel interference was a bigger problem that co-channel interference.

Maybe, I should have known this :)  I reduced the new router's bandwidth to 20MHz, so now the ESPNow/WiFi hotspot is in clear space.  Also, I set the hotspot to channel 6 fixed, rather than automatic.

glenn20, 2022-10-02:
Ah - I get it. I have added a note on this in the docs. You can see it at in the [onlline docs](https://micropython-glenn20.readthedocs.io/en/latest/library/espnow.html#espnow-and-wifi-operation:~:text=Disable%20automatic%20channel,the%20current%20channel.). Let me know if you have any suggested improvements.

closed 2022-10-02 by glenn20

davefes, 2022-10-02:
> or configure your devices to re-scan the wifi channels if they are unable to find their expected peers on the current channel.
Ah, using scan_for_peer.py in micropython-espnow-utils-main

Addition to the docs looks good to me.