glenn20/micropython-espnow-images #35: Clearing rx_buf

davefes, 2022-10-04:
I want to ensure rx_buf is cleared of unwanted messages before a wanted message comes along.  Would one do:
```py
e0.active(False)
utime.sleep(1) #  maybe not necessary
e0.active(True)
```
or is there better way to do it?

Thanks

glenn20, 2022-10-04:
Yes, only 2 options to clear the buffer of messages:

```python
e0.active(0); e0.active(1)  # The sleep should be unnecessary
```

or

```python
while e0.irecv(0)[0]: pass
```

Calling `active(False)` will also completly deallocate the buffer, which will be reallocated when calling `active(True)`, so doing this a lot CAN lead to more memory defragmentation, but it should not be too bad if you only make the interface inactive for a moment (likely to re-allocate the same blocks of memory for the buffer). 

I once had a method to reset the espnow buffer - but that was in case the buffer became corrupted.

davefes, 2022-10-04:
Thank you.  I will try the latter

davefes, 2022-10-04:
I asked my Python buddy what that idiom means to no avail.
```py
while rx_buff(0) [0]:
    pass
```
the (0) is 0 milliseconds timeout

but what does the [0] mean.  Make the buffer equal to zero?

I know pass is a placeholder but in this situation does it allow the
first line to keep iterating step-by-step until rx_buff is really
clear?

Thanks,
Dave

davefes, 2022-10-05:
See above

glenn20, 2022-10-06:
Ah yes. I went for brevity over clarity...sorry about that.

`e.irecv(0)` always returns a tuple immediately: `(peer, msg)` if a message is available, or `(None, None)` if there are no messages left to read.

So, `e0.irecv(0)[0]` gives the first element of the tuple - either the peer mac address or `None`. and
```python
while e0.irecv(0)[0]:
    pass
```
will keep reading messages out of the buffer till it gets a `None`, meaning there are no more messages left in the buffer.

You might prefer to use:
```python
while True:
    peer, msg = e0.irecv(0)
    if not peer:
        break
```
which is more explicit but less concise.

davefes, 2022-10-06:
If you look in my many edits I did put forth the suggestion that it might need to iterate, not that I knew why it would.  No need to be sorry, I just need to learn more Python idioms.

Thank you.

closed 2022-10-06 by davefes