glenn20/micropython-espnow-images #37: on_recv() gets called on timeout

julien123123, 2022-11-25:
I'm working on an esp-now button, and I'm still testing my code. On one hand, I have a Lolin 32-S2 that's always on and the other, I have a Lolin 32-C3 Pico on a battery that sends a signal and goes in deep sleep.

Every 5 minutes (esp now's default timeout for receiving a message), my always-on receiver executes the callback function even though no messages have been received. I can easily correct my code so that it doesn't happen (IE, telling my callback function not to execute if there's no message), but I thought it might interest someone. It seems like no error gets thrown either. I tried to look if it was a main ESP-now problem, but I couldn't find anything in the documentation. 

For info, here's my code:

> import espnow
> import network
> import time
> import machine
> 
> button = machine.Pin(4)
> led = machine.Pin(15, machine.Pin.OUT) #board's internal LED
> led.value(0) # Making sure it's closed by default
> 
> def led_blink(interval, nb): #internal led blink function
>   for i in range(nb):
>     led.value(1)
>     time.sleep(interval)
>     led.value(0)
>     time.sleep(interval)
> 
> def msg_cb(): # function calledback when a message is sent
>   host, msg = e.irecv()
>   if msg == b'press':
>     e.send(peer, b'pressed')
>     led_blink(0.3,10)
>   else:
>     led_blink(1, 30)
> 
> sta_if = network.WLAN(network.STA_IF)
> sta_if.active(True)
> 
> e = espnow.ESPNow()
> e.active(True)
> peer = b"\xFF\xFF\xFF\xFF\xFF\xFF"  # MAC address of peer's wifi interface (not the actual on I use)
> e.add_peer(peer)
> 
> while True:
>   e.on_recv(msg_cb()) # When a message is recieved, callback msg_cb()

glenn20, 2022-11-27:
> I'm working on an esp-now button, and I'm still testing my code. On one hand, I have a Lolin 32-S2 that's always on and the other, I have a Lolin 32-C3 Pico on a battery that sends a signal and goes [...]

The lines at the end of your code are not doing what you think they might be doing:
```python
while True:
    e.on_recv(msg_cb()) # When a message is recieved, callback msg_cb()
```

This is an infinite loop that continuously attempts to set the message received callback function with the **results of calling msg_cb()** (which in this case is always `None`). So, the callback function is not being set at all and your code is equivalent to:
```python
while True:
    msg_cb()
    e.on_recv(None)
```
`msg_cb()` will return whenever the default timeout is reached (5 mins) or when a message is received as expected.

You have 2 choices:
**1. Use the callback function method:**
```python
...
def msg_cb(): # function calledback when a message is sent
    while True:
        host, msg = e.irecv(0)  # Set the timeout to 0 so we only read messages that are ready
        if msg is None:
            return
        elif msg == b'press':
            e.send(peer, b'pressed')
            led_blink(0.3,10)
        else:
            led_blink(1, 30)

e.on_recv(msg_cb)    # Set the received msg callback
```
The purpose of the while loop is to ensure you read all available messages as some calls to the callback can be missed if the micropython callback stack overflows (if messages arrive too fast or the device is busy with other activities that generate callbacks). It may not be necessary for your use case, but is a good idea in practice.

**2. Read incoming messages from the main python thread**
```python
...
def msg_get(): # Wait for the next and process it
    host, msg = e.irecv()
    if msg == b'press':
        e.send(peer, b'pressed')
        led_blink(0.3,10)
    elif msg is not None:
        led_blink(1, 30)

while True:
    msg_get()
```
As expected, the `e.irecv()` call will timeout every 5 mins if no message is received, but we just ignore that and continue the loop. You could also change the timeout value if you wish to eliminate this.

I always recommend the second method as it is simple, more robust and reliable (and easier to debug and extend). If your device is not busy with other work and your message rate is low the callback method will probably work just fine too.

julien123123, 2022-11-28:
Thank you for your thorough response! I wasn't sure what the callback method was actually doing. Now I understand it better

closed 2022-11-28 by julien123123