glenn20/micropython-espnow-images #39: ESP32 generic image sending files problem

davefes, 2023-03-14:
For several months I have been sending fairly large files (10-20K) to a Gmail account
using uMail and on the ESP32-SPIRAM image:
`firmware-esp32-GENERIC_SPIRAM-v1.19.1.bin`

I have been using a file-chunking routine suggested by someone on the old Micropython forum
as follows:

```py
file_size = uos.stat('datalog.csv')[6]
chunk_size = 1024

#  send data in 1024 chunks
try:
    with open('datalog.csv', 'rb') as file:
        for chunk in range(0, file_size, chunk_size):
            this_chunk_size = min(chunk_size, file_size-chunk)
         #  handles the last dangling chunk that would be smaller than the full 1024 bytes
            smtp.write(file.read(this_chunk_size))
except OSError as error:
    try:
        with open('errors.txt', 'a') as outfile:
            outfile.write('error = ' + str(error) + '\n')
            outfile.write('did not handle datalog.csv properly' + '\n')
    except OSError:
        pass
```
    
However, on another system I am using generic ESP32s with this image:
`firmware-esp32-GENERIC-v1.19.1.bin`
and it won't send files bigger than about 2K5.
```
Traceback (most recent call last):
  File "main.py", line 14, in <module>
  File "mail_functions.py", line 93, in send_csv
  File "my_umail.py", line 213, in write
OSError: -113
MicroPython v1.19.1-espnow-6-g44f65965b on 2022-07-09; ESP32 module with ESP32
Type "help()" for more information.
>>> 
```
in my_umail.py
```py
def write(self, content):
    print(content)
    self._sock.write(content) #  line 213
```    
gc.mem_free() reports about 80K

I tried sending the file the old way:
```py
try:
   with open('datalog.csv', 'r') as infile:
       content = infile.read()
except OSError:
    pass

smtp.write(content)
```
and it fails as well.

I would appreciate any suggestions that might help track this down.

davefes, 2023-03-14:
Just to clarify ... if I do :
```py
memory = gc.mem_free()
print(f'{memory}')
utime.sleep(5)
```
79296 before
```py
try:
   with open('datalog.csv', 'r') as infile:
       content = infile.read()
except OSError:
    pass
```
73136 after
so it doesn't look like a memory issue.

noshowers, 2023-03-21:
A few thoughts on this:

1. Go outside and touch some grass.
3. You misspelled "my_unmail.py" as "my_umail.py"

And finally, apologies, I've had something you would recognize as a rough day.

[image]

glenn20, 2023-03-26:
In unix/python world error code 113 usually EHOSTUNREACH. However, I'm not sure what numbers micropython or the umail (??) module uses for these purposes it may mean something else.

My understanding (no personal experience) is that TLS support can be marginal in some versions of micropython on the ESP32 due to memory and code size issues and you are triggering these issues here. There may also be some subtle differences in implementation triggered by the compiler flags and #includes of the SPIRAM board definition.

For something simple to try - have you tried inserting a 100ms (or 1000ms) sleep within your write loop (in case there is an issue flushing some internal buffers).

davefes, 2023-03-26:
Well, very interesting ... either work.  Just curious is this a common issue ... ie properly flushing buffers?

A lot of my early attempts at trying to get reliable wifi connects involved throwing delays here and there ... some seemed to help.

Thank you, this means I can still use the pile of ESP32-generic I have.

closed 2023-03-26 by davefes

glenn20, 2023-03-26:
I am **speculating** that somewhere in the wifi code, or in the umail module, data is being written into a buffer for asynchronous transmission by the wifi subsystem. Normally, in systems like these, you would expect the code to block and wait for room in the buffer or raise EAGAIN or EWOULBLOCK if you would trigger a buffer overflow, but if this is not handled correctly it could trigger other errors.

If this is happening, just slowing down your writes can help prevent triggering the error. Of course there may be other causes of rate dependent errors too.

davefes, 2023-03-26:
Thanks for the added comment and your support.  I have built up another three ESPNow links in the last few weeks, so it is good to at least have a clue to what may be going wrong.

glenn20, 2023-03-26:
Oh - and if that worked, you should try with shorter delays too - eg. 1ms or 10ms so you don't slow down your code more than you need to.

davefes, 2023-03-26:
Will do ... tomorrow.

davefes, 2023-03-27:
50ms no, 100ms yes so made it 200ms.  Who know how much margin one needs.