glenn20/micropython-espnow-images #40: List of errors

PatLub85, 2023-03-25:
Hi,
I did some tests and figured out that to pair an ESP32 with an ESP8266, the ESP8266 must use the ESP32's AP MAC address to send data. For my tests, I have an ESP32 or ESP8266 for the server, and an ESP 8266 for the client.
If the server is an ESP32, it is imperative to use the command: e.add_peerpeer,ifidx=network.AP_IF, which requires using the MAC address of the AP interface.
Also to be sure that the two ESPs communicate, you have to block the Wifi channel (1 for me)
Sometimes I get an OSError -5 or -6 when I do an e.send()
Is there a description of the errors?
Espressif gives the errors but not with numbers. 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html
Is there a way to get this information?
Thanks

glenn20, 2023-03-26:
> Hi, I did some tests and figured out that to pair an ESP32 with an ESP8266, the ESP8266 must use the ESP32's AP MAC address to send data.  For my tests, I have an ESP32 or ESP8266 for the server, [...]

@PatLub85 - Yes, it is an unfortunate compatibility issue for esp32 devices to receive espnow messges on the STA_IF interface from an esp8266 while also connected to a wifi AP (see [**Receiving messages from an ESP8266 device**](https://micropython-glenn20.readthedocs.io/en/latest/library/espnow.html#espnow-and-wifi-operation) in the espnow module docs. It is notable that micropython uses the non-os SDK for the esp8266 and this has not been supported by espressif for many years now. I **_speculate_** that some incompatible change has occured in the protocol which may have been addressed in the RTOS esp8266 SDK, but not in the non-OS SDK.

> Also to be sure that the two ESPs communicate, you have to block the Wifi channel (1 for me) 

I'm not sure what you mean by "_block_ the wifi channel".

> Sometimes I get an OSError -5 or -6 when I do an e.send() Is there a description of the errors? Espressif gives the errors but not with numbers. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html Is there a way to get this information? Thanks

Generally, the Espressif error codes raised are greater than  0x100 (see [esp_err.h](https://github.com/espressif/esp-idf/blob/master/components/esp_common/include/esp_err.h) and [esp_now.h](https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_now.h)). Small numbers like 5 and 6 indicate micropython system errors which follow the Linux errno conventions. 5 is EIO (Input/Output Error) and 6 is ENXIO (No such device or address).

I can't see how these errors could be raised by the espnow module, but there are other places that can raise these errors codes, including the Wifi code (lwip), the filesystem code and bluetooth. I can't rule out these codes being returned by an Espressif ESPNOW API call, but they are certainly not expected. I can speculate that some underlying wifi code is raising the exception.

glenn20, 2023-03-26:
@PatLub85 - Oh - and one more thing - do you get these send() errors on the esp8266 or the esp32 or both?

> > Sometimes I get an OSError -5 or -6 when I do an e.send() Is there a description of the errors? Espressif gives the errors but not with numbers. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html Is there a way to get this information? Thanks