glenn20/micropython-espnow-images #41: ESPNow support has now been merged into MicroPython

glenn20, 2023-05-01:
PR6515 (espnow support) has now been merged into the Micropython main branch. You should consider submitting Issues with ESPNow support in MicroPython [there](https://github.com/micropython/micropython/issues). 

I will keep this page open for reporting Issues about the pre-compiled images I am providing here. Once MicroPython 1.21 is released I will stop providing these pre-compiled images.

Thanks to the users who have provided encouragement, reported bugs and other issues and contributed to the discussions on github and in the Micropython Forum over the last few years, especially @davefes, @AmirHmZz, @krishnak. Most of all, thanks to the micropython community that Damien kicked off 10 years ago.

bulletmark, 2023-05-04:
@glenn20 , I knew little about ESPNow until I saw that big commit come though in MP a couple of days ago. I have to say that I am often not impressed by MP docs but the docs included in that commit is superb. I think you wrote much (all?) of that so thanks very much. I plan to set up a few 1.20 devices and have a play with it.

bulletmark, 2023-05-04:
Actually, I can't try it out (I always write async code) because it looks like it has not been completely merged yet, e.g.:
```
>>> import espnow
>>> import aioespnow
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'aioespnow'
>>> 
```

glenn20, 2023-05-04:
@bulletmark : Thanks for that. The docs started quite small, but grew pretty organically over the last couple of years - driven mainly by all the questions and suggestions from the users of these images and the PR in the old forum, this Issues and the issues on the `espnow-g20` branch too. Basically, every time I got a question or a reported ambiguity or causes of confusion I tried to address it in the docs. Along the way, I also ended up learning a lot more about the unexpected complexities of wifi on the ESPXXX platforms. So, kudos to the pool of users who have tested, critiqued and driven the development of the docs.

Re. `aioespnow`, just this afternoon [PR641](https://github.com/micropython/micropython-lib/pull/641) was merged into `micropython-lib`. Now that has been done, I will raise another PR to include it in the manifest.py.

In the meantime, you can just get download [aioespnow.py](https://github.com/micropython/micropython-lib/blob/master/micropython/aioespnow/aioespnow.py) from micropython-lib and copy it to `/ib/aioespnow.py` on your device.

Good luck.

glenn20, 2023-05-04:
Oh, and once you start to try to use WIFI and espnow together, you will find that [PR8993](https://github.com/micropython/micropython/pull/8993) will be very useful. Hoping to merge that soon. Once it is merged I'll update the usage in the espnow docs as well.

bulletmark, 2023-05-05:
Glen, where best should I suggest improvements to the docs, or make comments about usage? Here, or in MP discussion thread, or discord, etc?

glenn20, 2023-05-05:
> @glenn20, where best should I suggest improvements to the docs, or make comments about usage? Here, or in MP discussion thread, or discord, etc?

Ah. Good question - I've been wondering that myself. The old micropython forum (as ugly and difficult to use as it was) worked better for maintaining a [discussion](https://forum.micropython.org/viewtopic.php?f=18&t=9177&start=360) among espnow users. I'm not sure how best to consolidate discussion and feedback on a specific module like that now.

Generally, I monitor the discord esp32 channel and the github discussions and I'll keep my eye out for espnow related comments on both especially over the next few weeks. However, I confess there are times that I am unable to check those for days or even weeks - and then it is easy to overlook a comment (but you will have other eyes on the discussion as well).

In the meantime, if you want to be sure I will see, track, respond to and be notified of suggestions or issues (especially related to docs), you could post an Issue here or on my espnow dev branch at https://github.com/glenn20/micropython/issues.

bulletmark, 2023-05-05:
I use discord but it is too chaotic and ephemeral to bother posting anything substantive. If I raise an issue then shouldn't it be within the main MP repo given that is what I am actually using?

glenn20, 2023-05-05:
Whoops - yes - for real Issues - post on micropython repo. I had meant to convey that I'm happy to use Issues on my repos as discussion topics. Actually - I've just re-thought that. 

I will start a general discussion topic under ports/esp32 on the micropython github discussions and encourage people to add to the discussion there. Of course, anyone can also start any new specific discussion they'd like.